timum-viw/socket.io-client #1: no matching function for call to 'SocketIoClient::on(const char [6], void (&)(uint8_t*, size_t))'

MrAta, 2016-07-18:
Hi, I'm using arduinoi IDE for windows and when i compile your basicExample get this error:
`C:\Program Files (x86)\Arduino\libraries\socket.io-client\examples\BasicExample\BasicExample.ino: In function 'void setup()':
BasicExample:39: error: no matching function for call to 'SocketIoClient::on(const char [6], void (&)(uint8_t*, size_t))'
     webSocket.on("event", event);
                                ^
C:\Program Files (x86)\Arduino\libraries\socket.io-client\examples\BasicExample\BasicExample.ino:39:32: note: candidate is:
In file included from C:\Program Files (x86)\Arduino\libraries\socket.io-client\examples\BasicExample\BasicExample.ino:7:0:
C:\Program Files (x86)\Arduino\libraries\socket.io-client/SocketIoClient.h:34:7: note: void SocketIoClient::on(const char*, std::function<void(const char*, unsigned int)>)
  void on(const char* event, std::function<void (const char * payload, size_t length)> func);
       ^
C:\Program Files (x86)\Arduino\libraries\socket.io-client/SocketIoClient.h:34:7: note:   no known conversion for argument 2 from 'void(uint8_t*, size_t) {aka void(unsigned char*, unsigned int)}' to 'std::function<void(const char*, unsigned int)>'
exit status 1
no matching function for call to 'SocketIoClient::on(const char [6], void (&)(uint8_t*, size_t))'`
what's the problem?

alxferraz, 2016-10-27:
Hey! change the function event to 

```
void event(const char * payload, size_t length) {
     USE_SERIAL.printf("got message: %s\n", payload);
}
```

the type of the first parameter is wrong in the example

timum-viw, 2016-12-07:
hey! fix it. Thanks!

closed 2016-12-07 by timum-viw