timum-viw/socket.io-client #2: Not able to connect to the socket server

angel1st, 2017-01-12:
Hi there,
First of all - it is really good having this port to socket.io, since that way we can keep server side code untouched.
Now back to the real issue I faced:
**Environment**
- Arduino IDE 1.8.0
- Adafruit ESP8266 breakout
- NodeJS server with running socket.io - it is tested and serving other socket.io clients, so it works just fine

*Note:* - it seems this Adruino has `lstdc++` added, since I don't get any link errors while building the code

**Libs & code**
I simply grabbed your lib, installed it and then try your *BasicExample*
Here is the exact code I have compiled:

    #include <Arduino.h>
    #include <ESP8266WiFi.h>
    #include <ESP8266WiFiMulti.h>
    #include <SocketIoClient.h>
    #define USE_SERIAL Serial

    ESP8266WiFiMulti WiFiMulti;
    SocketIoClient webSocket;

    void event(const char * payload, size_t length) {
      USE_SERIAL.printf("got message: %s\n", payload);
    }

    void setup() {
      USE_SERIAL.begin(115200);
      USE_SERIAL.setDebugOutput(true);
      USE_SERIAL.println();
      USE_SERIAL.println();
      USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
      }

      WiFiMulti.addAP("******", "*********");

      while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
      }

      webSocket.on("event", event);
      webSocket.begin("192.168.x.x", 4321);
    }

    void loop() {
      webSocket.loop();
    }

*Please Note*:
1. For security reasons SSID and its pass are masked, but I can see in the Serial Monitor output the connection is done successfully
1. 192.168.x.x is internal IP on my server and the port is 4321. Actually it was my only hesitance - since I have other socket.io client (from Angular based App), I am using the entire url, when connecting from them e.g. `http://192.168.x.x:4321`.

So when the app started, I can see successful connection to the network, the after couple of secs I see `[SIoC] Disconnected!` in the Serial Monitor console and this message is the only one after that.
It seems I cannot connect to the server.
Any ideas are appreciated,
Thanks,
Angel

timum-viw, 2017-01-13:
Hi Angel,

nice to see people can use this and  thanks for your cooperation.
Are you sure your other socket.io Clients are actually using websockets and are not using fall back to polling? The JavaScript socket.io client has this build in but it won't work for the ESP8266.
Cheers!

timum-viw, 2017-01-13:
But on a second look it is much more likely that the connection URL does not fit. Could you try to use a specific connection URL parameter?

angel1st, 2017-01-15:
Hi @timum-viw, thanks for your support. Actually after detailed research, I have decided to move to WebSocket and abandon Socket.io support, since the last one simply adds unnecessary over weight on top of the WebSockets.
Hence I didn't check last suggestion, however on the first look it makes sense.
Thanks anyways,
Angel

closed 2017-01-15 by timum-viw