timum-viw/socket.io-client #4: Can't emit

zenmanenergy, 2017-03-31:
I am able to connect to a node.js socket.io server through port 80. I am able to send events from the server to the esp8266, but I cannot seem to emit back to the server. 

It says that it's emitted, but nothing shows up in my server.

I am doing this:
webSocket.emit("whoami", "{ id: \"abc\"}");

It's printing this in the Serial monitor:

[SIoC] add packet 42["whoami",{ id: "abc" }]
[SIoC] packet "42["whoami",{ id: "abc"}]" emitted

AndresZeballos, 2017-05-02:
Hello @zenmanenergy 

I had the same problem until I changed for: webSocket.emit("whoami", "{ **"**id**"**: "abc"}");
Hope that help

timum-viw, 2017-05-03:
Yes, I'm quite sure the quotes are necessary. Does this close the issue?

timum-viw, 2017-05-03:
I'm going to update the readme accordingly

zenmanenergy, 2017-05-08:
Yes, that was it.

closed 2017-05-08 by zenmanenergy