timum-viw/socket.io-client #6: returns [SIoC] Disconnected!

nrk6208, 2017-06-03:
_**i am getting below error:**_

connected with Le-max2, channel 1
dhcp client start...
ip:192.168.43.115,mask:255.255.255.0,gw:192.168.43.1
[SETUP] BOOT WAIT 2...
[SETUP] BOOT WAIT 1...
[SIoC] Disconnected!
[SIoC] event disconnected not found. 1 events available
[SIoC] Disconnected!

**_here is the code:_**
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <SocketIoClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;
SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
    USE_SERIAL.begin(115200);

    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("Le-max2", "nrk@6208");

    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    webSocket.on("event", event);
    webSocket.begin("192.168.43.101",3484);
}

void loop() {
    webSocket.loop();
}

timum-viw, 2017-06-05:
hard to say. apparently your getting disconnected. Have you tried to connect to the server from a different machine? Does the server say anything about the attempt to connect?

nrk6208, 2017-06-13:
Thank god, anyway I know the issue, There is no error at client and server side. The main problem is with firewall. Once I have turned off the firewall, the client is connected with server.

closed 2017-06-13 by nrk6208

Bhupesh-V, 2019-02-21:
> Thank god, anyway I know the issue, There is no error at client and server side. The main problem is with firewall. Once I have turned off the firewall, the client is connected with server.

Can you please elaborate 
I am getting the same issue
[StackOverflow Issue](https://stackoverflow.com/questions/54800516/socketio-server-not-connecting-to-esp8266)