timum-viw/socket.io-client #7: Exception 28

gnkarn, 2017-06-23:
hi , i m experiencing this type of problem on my program running on node mcc 1.0
im using your code , as it has the capability to explicitly manage the different callbacks better than on the original library .

this is the error i see 

[SIoC] Connected to url: /socket.io/?transport=websocket
[SIoC] trigger event connect

Exception (28):
epc1=0x40207324 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff5e40 end: 3fff6300 offset: 01a0

if i comment out the web socket begin line in my code , the problem goes away, 
then i assume it may have to do with the connection process to the server .

the server , is a Heroku server .

i have added , Yield(), and delay  in the connect call back function , but with , no effect , i think that the code exits in the library , and it never reach, my callback function.

any hint to help my debugging is much appreciated ,

gnkarn, 2017-06-24:
more info
with the server disconnected , the error is
[SIoC] Disconnected!
[SIoC] trigger event disconnected

Exception (28):
epc1=0x40207346 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff5fb0 end: 3fff6330 offset: 01a0

with the server connected , the error is 
Port open
[SIoC] Connected to url: /socket.io/?transport=websocket
[SIoC] trigger event connect

Exception (28):
epc1=0x40207324 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff5e70 end: 3fff6330 offset: 01a0

timum-viw, 2017-06-25:
could you post your program code ? it seems like it has some problems finding the callbacks for connect and disconnect events. I have not tested the library with node MCU though..

gnkarn, 2017-06-26:
thanks for the reply

this is on the setup : 
 herokuWs.begin("app-gnk-p5js.herokuapp.com" , 80); // for socket.io  client library

	
	herokuWs.on("msgFromServer", wsVideoEvent);
	herokuWs.on("time", wsTimeEvent);
	herokuWs.on("connect", wsConnectEvent); 
	herokuWs.on("disconnected", wsDisconnectedEvent);

this is , in the loop

herokuWs.loop(); // handles Heroku socket.io loop
	herokuHeartBeat(); // send HB to Heroku WS server

and this on the callbacks

void  wsVideoEvent(const char * payload, size_t lenght) {
	// this to receive a message from the Heroku Server 
	USE_SERIAL.printf("[%u] Received from Heroku (video): %s\n", payload);
	String text = String((char *)&payload[0]);

	// if current efect is video continue else jump

	if (text.startsWith("Y")) {      // video message
		String xVal = (text.substring(text.indexOf("Y") + 1, text.length()));

	}
	//parse JSON objet

	// transfer objet to ledMatrix

	//end

	// send message to client
	// webSocket.sendTXT(num, "message here");

	// send data to all connected clients
	// webSocket.broadcastTXT("message here");

}
	
void  wsTimeEvent(const char * payload, size_t lenght) {
	yield();
	char text = payload[0];
	USE_SERIAL.printf("[%u] Received from Heroku (Time): %s\n", text);
}

void  wsConnectEvent(const char * payload, size_t lenght) {
	
	yield();
	char text = payload[0];
	USE_SERIAL.printf("[%u] Received from Heroku (connect): %s\n", text);
}

void wsDisconnectedEvent(const char * payload, size_t lenght) {
	char text = payload[0];
	USE_SERIAL.printf("[%u] disconnected from Heroku (disconnected): %s\n", text);
}

i understand i have to make proper treatment of the payload , it is working fine on my javascript client, but here i need to understand how to decode the json object passed by reference with the Json library ( still struggling ) with that one too

full code , is  here : https://github.com/gnkarn/Funky_Control8266-jqm/blob/Heroku%26p5js/Funky_Control8266-jqm/LEDWebsockets.h

timum-viw, 2017-08-03:
are you still having the problem? Did you try anything yet?

closed 2018-02-03 by timum-viw