timum-viw/socket.io-client #13: No matching function for call to 'SocketIoClient::emit(const char [17], String&)'

andagent, 2017-12-20:
Hello, I got an error while compiling simple sketch can you help me, pls?
Error: src\socket.cpp:13:42: error: no matching function for call to 'SocketIoClient::emit(const char [17], String&)'
```
#include <SocketIoClient.h>

SocketIoClient webSocket;
void setup(){

}
void loop(){
  String login = "test";
  String pswd ="demo";
  String code ="smth";
  webSocket.emit("login", login);
  webSocket.emit("pswd", pswd);
  webSocket.emit("statusCodeServer", code);
}
```

pawihd, 2018-01-26:
try both strings, like:
`webSocket.emit("event", "message§");`
and start then with:
`webSocket.begin("yourHostID",port);`

sarthakz9, 2018-01-29:
you can use 
`webSocket.emit("login", login.c_str());`

timum-viw, 2018-02-03:
This should answer the question. As stated in the readme the emit function expects two parameters of type ```char```. Your variables are of type ```String```. Use ```String::c_str ()``` to convert.

closed 2018-02-03 by timum-viw