timum-viw/socket.io-client #14: BasicExample does not run with my socket.io server!?

pawihd, 2018-01-25:
I try (desperately) to get communication from my ESP8266 board to my server (node.js/express/socket.io) but nothing works!?
This is my server:
```
var app = require('express')();
var http = require('http').Server(app).listen(3000);
var io = require('socket.io').listen(http);

io.on('connection',function(socket){
	console.log('a user connected!');
	socket.on('disconnect', function(){
              console.log('user disconnected');
	});
	socket.on('ESP', function(data){
             console.log('from ESP:',data);
             socket.emit('response', "Hi ESP);
	});
});

console.log("Socket-Server running @ http://localhost:" + "3000");

```
When i try to connect with proper SSD&Password and the following code in "BasicExample":
```
    webSocket.on("host", event);
    webSocket.begin("10.0.1.7:3000");
    webSocket.emit("ESP","Hi host!");

```
 i only get this response from ESP(Serial):
```
_[SETUP] BOOT WAIT 4...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 21
cnt 

connected with WBBA12, channel 11
dhcp client start...
ip:10.0.1.28,mask:255.255.255.0,gw:10.0.1.1
[SETUP] BOOT WAIT 3...
[SETUP] BOOT WAIT 2...
[SETUP] BOOT WAIT 1...
[SIoC] add packet 42["ESP",Hi host!]
[SIoC] Disconnected!
[SIoC] event disconnected not found. 1 events available
[SIoC] Disconnected!
[SIoC] event disconnected not found. 1 events available
... now forever!_

```
but no response on my server!

Questions:
1. where would i put in the port where the server listens in "BasicExample"?
I tried after the IP (see above!).

2. How would i send events? 
I tried "WebSocket.emit("event", msg)! 
Is this correct?

I appreciate very much any help!

timum-viw, 2018-01-25:
Hey there, as written in the readme you should give the port number as the second parameter to the connect function.
So in your case
```
webSocket.begin("10.0.1.7", 3000);
```

Let me know if this works!
Cheers

pawihd, 2018-01-26:
Great, thanks for the hint!

Yes, now i get a connection, at least!
And i even see that a message arrives at the ESP, but sending a message from there still fails!?
Obviously the webSocket.emit(„ESP“,“Hi host“); is not correct!?

—————————————————————————-
 webSocket.on(„host“, event); 		// <== works now!
 webSocket.begin(„10.0.1.7",3000); 	
 webSocket.emit(„ESP“,“Hi host“); 		// <== no event on server!?
—————————————————————————-

Arduino shows the keyword „emit“ not in red but in black. Does this mean, that the syntax is wrong?
But strangely enough, it compiles without an error!

This is what ESP prints out:

—————————————————————————-
Starting Wifi connection...
Waiting for a connection..

[SIoC] add packet 42["ESP",Hi host]
[SIoC] Connected to url: /socket.io/?transport=websocket
[SIoC] packet „42[„ESP“,Hi host]“ emitted    //is it really emitted?
[SIoC] event connect not found. 1 events available
[SIoC] trigger event host
event occured
got message: Hi Client  		// <== sent out by the server!
—————————————————————————-

What kind of message is this: „42[„ESP“,Hi host]“.

I tried to receive that on the server with:

—————————————————————————-
io.on('connection',function(socket){
	console.log(„a user connected“);	//<== do see this!
	socket.on('ESP', function(data){
    		console.log(‚ESP:‘,data);                 //<== did not see this log on server!
    		socket.emit(‚host‘,“received your message!“);
	});
});
—————————————————————————-

Do you know what's wrong?

Greetings
Peter

pawihd, 2018-01-26:
I changed:
`webSocket.emit(„ESP“,“Hi host“);`
to:
`webSocket.emit(„ESP“,“\"Hi server\"");`
and this does the job!!

(I have no idea, why, but It is obviously always good to read the readme!! ;-)
Thanks again for the help!

timum-viw, 2018-01-26:
I was about to suggest that. Has to do with the Socket.io protocoll. Glad to hear everything worked out!

closed 2018-01-26 by timum-viw

pawihd, 2018-01-26:
It took me now nearly a week to find a working example!
To find out, what really goes on, it would be very helpful to see what is really happening in the TCP packets!
Is there any (simple) tool for monitoring the net traffic?

Bhupesh-V, 2019-02-26:
hey Guys I need help with the similar problem

here is my sample Socket server running on localhost
```js
const io = socket.listen(server);
io.on("connection", (socket) => {
    console.log('made socket connection', socket.id);
    socket.on("ESP", function(data){
        io.sockets.emit('Hi ESP, ESP called', data);
        console.log("Socket Working !");
    });
    console.log('made socket connection', socket.id);
    socket.on("connect", function(data){
        io.sockets.emit('Hi ESP connect called', data);
        console.log("Socket Working !");
    });
    console.log('made socket connection', socket.id);
    socket.on("event", function(data){
        io.sockets.emit('first', data);
        console.log("Socket Working !");
    });
});
```

Here is my Client Code

```cpp
SocketIoClient webSocket;

const char* ssid     = "DyfoAirtel";      // SSID
const char* password = "Dyfo1234";        // Password
const char* host = "192.168.1.37";        // Server IP (localhost)
const int   port = 8080;                  // Server Port
const char* url = "http://192.168.1.37:8080/registerslavepi";

bool is_socket_connected = true;

char* string2char(String command){
    if(command.length()!=0){
        char *p = const_cast<char*>(command.c_str());
        return p;
    }
}

void make_socket_call(char* emit_key, String data) {
  // makes socket call.
  char charBuf[data.length() + 1];
  webSocket.emit(emit_key, string2char(data));
}

void on_disconnect(const char * payload, size_t length) {
  // calls on_disonnect of web socket
  Serial.println("Disconnected to Remote Server.");
  is_socket_connected = false;
}

void on_connect(const char * payload, size_t length) {
  // called when websocket connects.
  Serial.println("Connected to Remote Server.");
  String res_data = "{\"astro_id\": + \"34\"}";
  make_socket_call("jsonObject", res_data); // sens this key and value.
  if (is_socket_connected == false) { // if this connection is false,
    //send_initial_pins_status(); // sync the pin status to master.
    is_socket_connected = true;// now the connection is up ! hurray !
  }
}

void connect_websocket_events() {
  // only two events are handled: on_connect() and on_disconnect()
  Serial.println("Initiating socket bindings.");
 
  webSocket.on("connect", on_connect);
  webSocket.on("disconnected", on_disconnect);
  webSocket.begin("http://192.168.1.37", 8080, "/socket.io/?transport=websocket");
  webSocket.emit("ESP","\"Hi host!\"");
}

void setup(){
  Serial.begin(115200);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);     // 5ms Delay
    Serial.print(".");
  }

  Serial.print("");
  Serial.println("WiFi connected");
  Serial.println("IP address: " + WiFi.localIP());
  connect_websocket_events();
}

void loop(){     
     while(WiFi.status() == WL_CONNECTED){
      webSocket.loop();
      delay(1000);
      Serial.println("Retrying ....");
    }
}
```

I am getting same output as in the issue
```
18:13:46.220 -> ...............WiFi connected
18:13:53.253 -> ⸮
18:13:53.253 -> Initiating socket bindings.
18:13:53.253 -> [SIoC] add packet 42["ESP","Hi host!"]
18:13:53.253 -> [SIoC] Disconnected!
18:13:53.253 -> [SIoC] trigger event disconnected
18:13:53.253 -> Disconnected to Remote Server.
18:13:54.246 -> Retrying ....
18:13:54.246 -> [SIoC] Disconnected!
18:13:54.280 -> [SIoC] trigger event disconnected
18:13:54.280 -> Disconnected to Remote Server.
18:13:55.241 -> Retrying ....
18:13:55.275 -> [SIoC] Disconnected!
18:13:55.275 -> [SIoC] trigger event disconnected
18:13:55.275 -> Disconnected to Remote Server.
18:13:56.269 -> Retrying ....

```

Ahmedesayed, 2019-05-30:
> Great, thanks for the hint! Yes, now i get a connection, at least! And i even see that a message arrives at the ESP, but sending a message from there still fails!? Obviously the [...]

could you tell me how to recieve event data from server ??