timum-viw/socket.io-client #15: ESP32 compile issue

Kadah, 2018-01-29:
Using bbx10's ESP32 arduinoWebSockets (and its prerequisite Hash_tng) to get this to work.
https://github.com/bbx10/arduinoWebSockets/tree/esp32
https://github.com/bbx10/Hash_tng

The only issue I've hit now is that ESP32 lacks hexdump used at https://github.com/timum-viw/socket.io-client/blob/master/SocketIoClient.cpp#L41
I've commented this out in my local machine.

I had needed binary support on my project, it would have been convenient if this library supported those, but have since respec'd our comms to not require binary socket.io payloads. Of the few socket.io libraries I've gotten to work, I'm preferring the usage of this one more.

timum-viw, 2018-02-03:
Obviously the ```hexdump``` is merely a placeholder and doesn't do anything about actually handling the packet.
Would be nice if you could implement something and then open a pull request 😊

coma0815, 2018-06-04:
Is there any news about the esp32 and this library? @Kadah would you share your experiences? Thank you!

Bmooij, 2018-10-23:
@timum-viw what is the purpose of the function ```hexdump```?

jslay88, 2018-10-31:
Just comment out `hexdump` (its just a debug as far as i can tell) for ESP32.

I have a working Socket.IO client on my ESP32 with just installing this library and WebSockets library, and modifying this library to comment out `hexdump`.

yanoom, 2019-05-25:
Any news about compiling and running this project on ESP32?
Would it be possible for someone who succeeded to fork a separate project (/explain how it's done)?

10x