timum-viw/socket.io-client #16: Delayed successful SSL connection with node.js server and Esp8266 client

kapoorsarthak, 2018-02-01:
i am using a node.js server to communicate with my ESP8266 client using SSL but the connection is not stable and it takes 8 to 9 tries sometimes and sometimes 20-25 tries to connect to the server the number of tries are not fixed at all. the SSL certificate uses a 2048 bit key. Using the library websocket

##Basic Infos
Hardware
Hardware:	ESP-01

##Description
Connection not stable

##Settings in IDE
Module: Generic ESP8266 Module

##Sketch
#include <WebSocketsClient.h>
WebSocketsClient webSocket;
WebSocket.beginSSL("hostname.com",port,"/socket.io/transport=websocket","fingerprint");

timum-viw, 2018-02-03:
I think this is either an issue with your server or the WebSockets library. So could you head over to their issue section and open a request there?

kapoorsarthak, 2018-02-03:
@timum-viw  I am using the socket-io.client library created by you ( that uses websocket internally ), and is it possible to do ssl with ESP8266 - 01 , or is there any memory constraint.

timum-viw, 2018-11-05:
this should be fixed.

closed 2018-11-05 by timum-viw