timum-viw/socket.io-client #21: Add lifecycle handlers (connect, disconnect, etc)

suyashkumar, 2018-03-06:
I propose that user-supplied lifecycle functions should be called on useful events like socket `CONNECT` and `DISCONNECT` events. This will allow users of this library to execute any application logic necessary when these events occur. 

For example, when the socketio client disconnects and reconnects, there may be some application logic needed to reset state or renegotiate with the server (there will be a new session id issued upon the new connection as well). 

I can go ahead and put a PR up for this soon--let me know if this is something that makes sense to you!

timum-viw, 2018-05-06:
Actually this is already implemented by emitting events 'connect' and 'disconnect' which you can subscribe to. Is that what you had in mind?

suyashkumar, 2018-07-28:
Indeed! Ended up going that route, thanks @timum-viw!

closed 2018-07-28 by suyashkumar