timum-viw/socket.io-client #22: oncatenate variable int in emit

OmarGC, 2018-05-06:
How to concatenate variable double or int to socket.emit ("event name", "\" CHAIN ​​TO SEND \ "")

Achieve send the normal chain, but the ideal is to send the variable that comes from the sensor.
the question is that I put the structure + String (dataInteger) +
but that sends me error.

example

    `int dis = 88;`
    `webSocket.emit ( "Esp8866_1" , "\" Cisterna: "+String(dis)+"\" " );`
    `webSocket.begin ( "192.168.1.11" , 3000 , "/socket.io/?transport=websocket" );`

OmarGC, 2018-05-06:
Then I converted to char and sent me this error.
what I want is to send sensor variable by emit

`
/home/omargc/Escritorio/NodeMCU/funcinales/SocketIO_BasicExample/SocketIO_BasicExample.ino: In function 'void setup()':
SocketIO_BasicExample:46: error: invalid operands of types 'const char*' and 'const char [3]' to binary 'operator+'
     webSocket.emit ( "Esp8866_1" , "\" Cisterna: "+char(dis)+"\" " );
                                                              ^
exit status 1
invalid operands of types 'const char*' and 'const char [3]' to binary 'operator+'`

timum-viw, 2018-05-06:
The concatenation results in a String object. You have to convert it to char *.
Try
`webSocket.emit ( ("Esp8866_1" , "\" Cisterna: "+String(dis)+"\" ").c_str() );`

OmarGC, 2018-05-06:
Perfect
now he does not send me error
Thank you

timum-viw, 2018-05-06:
You're welcome!

closed 2018-05-06 by timum-viw