timum-viw/socket.io-client #23: Emitting data, event not recognized in NodeJS SocketIO

ngfelixl, 2018-05-06:
Hi,
first of all great work so far. I've got stuck on a problem with retrieving emitted events in Nodejs. It recognizes the esp8266 and registers the esp as socket client successfully. The esp is connected via Wifi to the same network as my pc is connected to. But I can't get data transmission working.

My **Node** code looks like
```
[...]
const app = express();
const port = 81;

const server = app.listen(port, '192.168.x.x', () => { console.log('SocketIO port ' + port); });
const io = new socketio(server, {'origins': '*:*'});

io.on("left", (data: any) => {
  console.log('Esp emit: ', data);
});
```
In the **esp8266** sketch I'm emitting like.
```
setup() {
  [...]
  webSocket.begin("192.168.x.x", 81);
}
void loop() {
  [...]
  webSocket.emit("left");
}
```

**Serial** output prints
```
[SIoC] add packet 42["left"]
[SIoC] packet "42["left"]" emitted
```

Node Socket interaction with angular serving webserver works perfectly. Any suggestions? Thanks for your help.

ngfelixl, 2018-05-06:
It's working with default websocket instead of socketio. Are there any suggestions for using socketio directly?

closed 2018-05-06 by ngfelixl

timum-viw, 2018-05-06:
Hey there, "default websocket instead of socketio" for server or client?

ngfelixl, 2018-05-06:
Using default websocket for node server. Received data then is "42["left"]".
Got socketio now working with another socketio library https://github.com/robojay/Socket.io-v1.x-Library. But would like to use your library as it has a cleaner interface, I think.