timum-viw/socket.io-client #27: Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads

Tvde1, 2018-06-06:
I'm getting this error whenever my esp calls `webSocket.begin()`.
Here's my sketch:

```
#include <Arduino.h>

#include <ESP8266WiFi.h>

#include <SocketIoClient.h>

#define USE_SERIAL Serial

SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
    USE_SERIAL.begin(115200);

    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFi.begin("ssid", "pass") //Edited these out;

    while(WiFi.status() != WL_CONNECTED) {
    	Serial.print(".");
        delay(100);
    }

    Serial.println();
    Serial.println("Connected.");

    webSocket.on("abc", event);
    webSocket.on("disconnected", event);
    webSocket.begin("192.168.2.5", 80);
}

void loop() {
    webSocket.loop();
    delay(1000);
}
```
and here is my nodejs server:
```
const express = require('express');
const http = require('http');
const server = http.createServer()
const socketio = require('socket.io');

const app = express(server);

app.on('/', (req, res) => {
    res.send(200, 'text/plain', 'Ok');
});

const io = socketio(server);

io.on('connection', socket => {
    console.log('Connection.');
    socket.emit('abc');
});

app.listen(80);
```
I have no idea why it would do this. Is this lib outdated and won't it work with socket.io anymore?

Tvde1, 2018-06-06:
This happens both when my server is off and on. It also doesn't log a new connection.

rremy, 2018-07-03:
This happens because the "connect" and the "disconnect" event called with null pointer, and you try to print the null pointer in USE_SERIAL.printf 

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

timum-viw, 2018-09-12:
rremy is right. does that fix your problem?

Tvde1, 2018-09-12:
I am unable to try now, but I will retry and reopen the issue if it persists.

closed 2018-09-12 by Tvde1