timum-viw/socket.io-client #30: Use root CAs to verify SSL certificates?

suyashkumar, 2018-07-28:
Hi @timum-viw, I noticed that currently `beginSSL` requires a fingerprint of the server SSL certificate for verification. Since those host SSL certs can change often (every 90 days for the popular LetsEncrypt certs), it seems that it may be beneficial to leverage _root_ CA certificates to perform the verification of the host SSL certificate as done in this [example from Arduino's WiFiClientSecure](https://github.com/esp8266/Arduino/pull/3271/files). 

I know that this change should probably just be made in the downstream websockets library (I opened up an issue for that at Links2004/arduinoWebSockets#352) but figured I'd open an issue here for discussion and to see if you or other folks using this library have run into this issue of re-flashing firmware everytime a host SSL certificate changes. 

If this is an issue other folks are facing as well, I'd certainly be willing to pitch in to help solve this problem instead of me just keeping it on the backburner as well. Also feel free to close if you don't think this is relevant. Thanks!

timum-viw, 2018-09-12:
Hey there! Actually I myself have never used SSL with this library as I never had security sensitive applications. But your request sounds reasonable. So I'd be happy to take action once something comes up concerning this library. Sorry I won't be more of a help than that 😄 Cheers!

sarthakz9, 2018-09-30:
> If this is an issue other folks are facing as well

yes, we are. Did you find a solution for this? @suyashkumar

suyashkumar, 2018-10-01:
@sarthakz9 yep, this can be fixed with some changes in the downstream web sockets library (see Links2004/arduinoWebSockets#352). I pretty much know what needs to be done, but I haven't had a chance to make a PR. I'll probably see if I can circle back to it this weekend! Let me know if you'd like to help with a PR!

sarthakz9, 2018-10-01:
sure, I would like to help. 💯

timum-viw, 2018-11-05:
Any news on this? Do we have to change something in this library to make it work?

Muzzamil1, 2020-02-22:
Hi @suyashkumar, were you able to solve it?

closed 2021-01-13 by timum-viw