timum-viw/socket.io-client #34: Namespace support

jslay88, 2018-11-01:
Would be really nice if we could get some namespace support.

I will try and take a look at the Socket.IO spec and see if I can't manage something myself, but I am not a strong C/Arduino coder.

mawildoer, 2018-12-13:
Hi @jslay88,

Have you made any progress on this one? Happy to help, but having trouble finding the relevant sections of the socket.io docs/code

"doConnect" in [here (socket.io repo)](https://github.com/socketio/socket.io/blob/e0b2cb0c5a9af768a48c43f65643412cc8edc7df/lib/client.js) seems to be promising, but doesn't really contain the information as to when it's called

jslay88, 2018-12-13:
it is called on connect.

```
Client.prototype.connect = function(name, query){
  if (this.server.nsps[name]) {
    debug('connecting to namespace %s', name);
    return this.doConnect(name, query);
  }

  this.server.checkNamespace(name, query, (dynamicNsp) => {
    if (dynamicNsp) {
      debug('dynamic namespace %s was created', dynamicNsp.name);
      this.doConnect(name, query);
    } else {
      debug('creation of namespace %s was denied', name);
      this.packet({ type: parser.ERROR, nsp: name, data: 'Invalid namespace' });
    }
  });
};
```

Which I have to assume is the function you are calling when are calling `io.connect()`

Problem is, I am not the strongest JavaScript programmer, and I am having trouble following the code (particularly, where is `this.server.checkNamespace` server class actually defined).

mawildoer, 2018-12-13:
In some experimenting, I found the request made here:
(where the namespace is 'youll notice this'

[image]

mawildoer, 2018-12-13:
This contains a bit more information regarding the encoding scheme, but honestly raises more questions than is answers: https://github.com/socketio/engine.io-parser/blob/master/lib/index.js

mawildoer, 2018-12-13:
Okay, so based on:
https://github.com/socketio/engine.io-parser/blob/master/lib/index.js

Lines >15
```
/**
 * Packet types.
 */

var packets = exports.packets = {
    open:     0    // non-ws
  , close:    1    // non-ws
  , ping:     2
  , pong:     3
  , message:  4
  , upgrade:  5
  , noop:     6
};

var packetslist = keys(packets);

/**
 * Premade error packet.
 */

var err = { type: 'error', data: 'parser error' };

/**
 * Encodes a packet.
 *
 *     <packet type id> [ <data> ]
 *
 * Example:
 *
 *     5hello world
 *     3
 *     4
 *
 * Binary is encoded in an identical principle
 *
 * @api private
 */
```

>196
```
 * Encodes multiple messages (payload).
 *
 *     <length>:data
 *
 * Example:
 *
 *     11:hello world2:hi
```

I've worked out that the above message in the request payload (last post) is 21 characters long, which is the first specifier (lines 196 onward) and the 4 in 40 signifies a message. 

[image]

Specfically, according to the above, a connection message (the zero is processed elsewhere)

nis267, 2020-07-24:
Hello, someone knows how to connect to a namespace with this library or knows another library with which it's working ?

Thank you

coma0815, 2020-07-24:
https://github.com/timum-viw/socket.io-client/pull/81
There is a pull request - try to merge it in your library!

nis267, 2020-07-24:
Thank you @coma0815 for the fast answer it's working !

closed 2021-01-13 by timum-viw