timum-viw/socket.io-client #36: OTA update support

jslay88, 2018-11-01:
I think it should be possible to perform an OTA update using SocketIO binary transport.

Would this be something that could be implemented or investigated? Would be nice to send a FW updates to the ESP devices if they are out of date.

timum-viw, 2018-11-05:
puh, that would be quite the effort and actually I think that would be more something for another library as it has nothing to do with socket.io. This library would receive the FW through the binary transport as you suggested  and then you pass it on to some other library.

jslay88, 2018-12-13:
I'm more interested in how it would work (as I am deficient in Arduino) more so that building support into this library. Having said that, if its fairly straight forward enough (not too bloated of a process), I think it would be somewhat useful within this library since it specifically targets ESP devices.

closed 2021-01-13 by timum-viw