timum-viw/socket.io-client #41: send emit in void loop

bangjii, 2018-11-09:
hay its really cool to made this library btw, its really usefull to connect my esp8266 to my socketio server.
i have a question, why there's nothing happen when i emit every 5 second in void loop()? i just using simple code:
`socket.emit("sensor-id", "{\"id\":\"mesinB\"}");`
in serial monitor:
[SIoC] packet "42["sensor-id",{"id":"mesinB"}]" emitted
but nothing show in my socketio server.
but if that code (`socket.emit("sensor-id", "{\"id\":\"mesinB\"}");`) i put in void setup() just one time, my server read that json text successfuly.
sorry my bad english

timum-viw, 2019-01-16:
do you still have that problem? Could you show some more code?

Bhupesh-V, 2019-02-21:
> hay its really cool to made this library btw, its really usefull to connect my esp8266 to my socketio server.
> i have a question, why there's nothing happen when i emit every 5 second in void loop()? [...]

I tried this thing It's not working for me :cry: 
Any suggestions ?

laisangbum, 2020-08-15:
You just put the websocket.loop after  the websocket.emit in void loop(); I tested and it success to send message to server.

closed 2021-01-13 by timum-viw