timum-viw/socket.io-client #47: socket.on("disconnect", handler) not working

RobertHolscher, 2019-01-23:
In the documentation you write that the disconnect event is triggered when the connection is lost with the server. I am trying to use that with the arduino sketch but the compiler says:
No matching function for call to 'SocketIoClient::on(const char [11], void(&)())'

The call I am using is webSocket.on("disconnect", disconnectevent);

The disconnectevent is the handler 
Am I using it wrongly ?

coma0815, 2019-01-23:
Do you have a matching function called "disconnectevent"? 
In my sketch it would look like this:

`void disconnected(const char * payload, size_t length) {
    // do something when disconnected....
}

webSocket.on("disconnect", disconnected);`

RobertHolscher, 2019-01-24:
Thank you, forgot to put in the (const char * payload, size_t length), overlooked that. Thanks for your reply

RobertHolscher, 2019-01-24:
It compiles ok now , case closed

closed 2019-01-24 by RobertHolscher