timum-viw/socket.io-client #48: invalid conversion from 'int' to 'const char*'

ClaintEastwood, 2019-02-19:
Hello! first thanks for your work, I have a problem, I am doing a project with arduino and dht11 to send a temperature to the nodejs server:

**int temperature;**

And I want to send with:

**webSocket.emit (temperature);**

But it causes an error: invalid conversion from **'int' to 'const char *' [-fpermissive]**

How can I fix it? or how can I send the variable temperature to the nodejs server?

Thank you very much

Jefinho85, 2019-02-19:
Hi, you can post your complete code?

Jefinho85, 2019-02-19:
For your information, the library says the following:
`void emit (const char * event, const char * payload = NULL);`

That is, to send something you need to say where you want to send and WHAT you want to send.
ex:
`webSocket.emit ("myTopic", floatTempC);`

If you post your complete code, more elaborate help may be offered.

Cheers!

ClaintEastwood, 2019-02-19:
Hi Jefinho85, I post:

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <DHT.h>
#include <SocketIoClient.h>

#define USE_SERIAL Serial

int temperature;
int SENSOR = 2;
DHT dht (SENSOR, DHT11);

ESP8266WiFiMulti WiFiMulti;
SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
    USE_SERIAL.begin(115200);
    dht.begin();

    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("JAZZTEL_mdRA", "WIFI@jazz1717@");

    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    webSocket.on("event", event);
    webSocket.begin("192.168.1.220", 1234);
    // use HTTP Basic Authorization this is optional remove if not needed
    //webSocket.setAuthorization("username", "password");
}

void loop() {
  temperature = dht.readTemperature();
  Serial.print("Temperature: ");
  Serial.print(temperature);
  
    webSocket.emit(temperature);
    webSocket.loop();
}
```

and error:

> ..../sketch_feb19a.ino: In function 'void loop()':
> sketch_feb19a:53:31: error: invalid conversion from 'int' to 'const char*' [-fpermissive]
>      webSocket.emit(temperature);
>                                ^
> .../sketch_feb19a.ino:5:0:
> .../Arduino/libraries/Socket.io-v1.x-Library-master/SocketIoClient.h:42:7: error:   initializing argument 1 of 'void SocketIoClient::emit(const char*, const char*)' [-fpermissive]
>   void emit(const char* event, const char * payload = NULL);
>        ^
> exit status 1
> invalid conversion from 'int' to 'const char*' [-fpermissive]
> 

Thanks!

Jefinho85, 2019-02-19:
The second argument of the emit function requires a `const char * payload = NULL`. The temperature read by a `DHT `is given in `float`, ie you need to convert `float `to `char`. You'll need to know about `dtostrf`. Another detail, its temperature statement as int is wrong, because `DHT `requires `float`. So is the correct one:
```
#define DHTPIN D7 //
#define DHTTYPE DHT11 // DHT 11
DHT dht (DHTPIN, DHTTYPE);
```
Your loop never waits a bit, that's a problem. You need to take a long time to read the `DHT`, then to the end of the use `delay (1000)`.

Look at the `DHT `example in the arduino IDE and then look at the `SocketIO `example in the arduino IDE. Put the two together and it will work. Just be sure to use dtostrf function to bring temperature flot to char. :)

ClaintEastwood, 2019-02-19:
It works! you are my hero :P

Thank you again

closed 2019-02-19 by ClaintEastwood