timum-viw/socket.io-client #50: Conectar a un dominio https://dominio.com

OmarGC, 2019-03-17:
I want to connect my esp8266 with heroku. My problem is the connection with https since I use the library with http putting my ip public and it works

To start the connection I put these lines of code:
io.begin ( "https://dominio.herokuapp.com", 80, "/socket.io/?transport=websocket" );

Through the serial port sends me this message:
[SIoC] Disconnected!
[SIoC] event disconnected not found. 1 events available.

Could you tell me an example of how to connect with https domains?

elastra21, 2019-04-04:
Did you find the error?, i am stuck in the same situation

timum-viw, 2019-04-06:
Https means your server uses ssl. You have to use the SocketIO.beginSSL function open the connection.

rishabhrpg, 2019-04-23:
.beginSSL function requires fingerprint parameter how do I get fingerprint?

OmarGC, 2019-04-28:
> Https significa que su servidor utiliza ssl. Tienes que usar la función SocketIO.beginSSL para abrir la conexión.
Trying to solve e using the socket.beginSSL method.
But it does not work because in the documentation I see that I need a parameter: fingerprint. Can you get this heroku service parameter?
I have a home configuration without https and I'm fine, but it does not have an ssl certificate. My question is when do I get that certificate, how can I connect it? Or where do I get the fingerprint?
I read that you get it when you buy your SSL certificate. I am new at this. Thank you

OmarGC, 2019-07-07:
I keep trying...
Obtain a certificate with certboot and my home site has https, then with the command:
sudo openssl x509 -noout -in /etc/letsencrypt/archive/smartliquid.org/cert1.pem
I get a fingerprint.

now I connect with beginSSL and send me the same error, someone to solve?

Muzzamil1, 2019-07-19:
Hi OmarGC  
Did you solve it ?

Chevindu, 2019-10-09:
> Or where do I get the fingerprint?

@OmarGC Did you try your Heroku URL with [this one](https://www.grc.com/fingerprints.htm)?

raulest50, 2019-12-28:
I'm having the same problem, in localhost:3000 works perfect but im unable to get it work in online server. I tried in heroku and glitch but no luck. This is what my nodeMCU prints when i use :
webSocket.beginSSL("https://maj****.glitch.me", 80, "/socket.io/?transport=websocket", "8E 68 78 DD 47 8F 03 BF 8B 51 DB 62 FE 5A CB 6E E9 20 82 A1");

 i get this output from serial port debug ( tried with http:// https:// ws://) :
[WS-Client] connect wss...
[WS-Client] connection to https://majoma-smart.glitch.me:80 Failed
[WS-Client] client disconnected.
[SIoC] Disconnected!
[WS-Client] connect wss...
[WS-Client] connection to https://majoma-smart.glitch.me:80 Failed
[WS-Client] client disconnected.
[SIoC] Disconnected! ...

i used the fingerprint from the url that @Chevindu posted, with : and blank spaces without achieving connection.

when i use begin only specifing the port 80 like this:
webSocket.begin("https://majoma-smart.glitch.me", 80);

i get this serial output (also tried with http:// https:// ws://):

WiFi connected
192.168.1.64
[WS-Client] connect ws...
[WS-Client] connected to https://maj******.glitch.me:80.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] handshake GET /socket.io/?transport=websocket HTTP/1.1
Host: https://maj*****.glitch.me:80
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: jz0RkVKI1fAzS8pUOypgwA==
Sec-WebSocket-Protocol: arduino
Origin: file://
User-Agent: arduino-WebSocket-Client

[write] n: 290 t: 3694
[WS-Client][sendHeader] sending header... Done (336433us).
[WS-Client] connection lost.
[WS-Client] client disconnected.
[SIoC] Disconnected!
[WS-Client] connect ws...
[WS-Client] connected to https://maj*****.glitch.me:80.
[WS-Client][sendHeader] sending header.......

it seems that reach the server but for some reason cant complete the connection. at least the printed info gives more hopes than using beingSSL. 

already checked port at server side and everything is correct. Even i got it to work perfectly using this python3 script:

import socketio # python-socketio

sio = socketio.Client()

salida = 0

@sio.event
def connect():
    print('connection established')

@sio.event
def disconnect():
    print('disconnected from server')

@sio.event # custom event
def identificarse(data):
    print('identificandose ', data)
    sio.emit('identificacion', 'nodo1')

sio.connect('http://maj******.glitch.me:80')
sio.wait()

it would be nice to know if the author have tested this with online server or just locally, just to know if there is any hope or better start looking for other solutions.

nevertheless thanks for this lib @timum-viw very nice work.

Chevindu, 2019-12-28:
> It seems that reach the server but for some reason cant complete the connection. at least the printed info gives more hopes than using beingSSL.
> 
> Already checked port at server side and everything is correct. 

It seems this is due to how Heroku handles web-apps at runtime. I don't think I understand this well enough to explain, because I come from a mechanical background rather than computer science. Please [refer to this answer](https://stackoverflow.com/a/51572239) for more info.

I switched from Heroku to AWS EC2 and never looked back. Good luck! 

Thanks @timum-viw for this easy to use lib.

raulest50, 2019-12-28:
thank you very much @Chevindu  for your interest, im going to try with AWS EC2. in case of success i will get back here.

Sakthivelgovindan, 2020-08-30:
Hi @OmarGC, @elastra21, @Chevindu , @raulest50, @Muzzamil1 , @rishabhrpg  and @timum-viw ,
I tried to connect using the below code, It's working fine

Example :  
       webSocket.begin("demo.herokuapp.com");

no need to use webSocket.beginSSL, no need to use webSocket.begin(https://iotbackenddemo.herokuapp.com) with http or https and no need to provide port number, the above code is working.

Sakthivelgovindan, 2020-08-30:
Hi @OmarGC, @elastra21, @Chevindu , @raulest50, @Muzzamil1 , @rishabhrpg and @timum-viw ,
I tried to connect using the below code, It's working fine

Example :
webSocket.begin("demo.herokuapp.com");

no need to use webSocket.beginSSL, no need to use webSocket.begin(https://demo.herokuapp.com) with http or https and no need to provide port number, the above code is working.

raulest50, 2020-08-30:
> Hi @OmarGC, @elastra21, @Chevindu , @raulest50, @Muzzamil1 , @rishabhrpg and @timum-viw ,
> I tried to connect using the below code, It's working fine
> 
> Example :
> 
> #include <SocketIoClient.h>
> SocketIoClient webSocket;
> 
> void setup(){
> webSocket.begin("demo.herokuapp.com");
> }
> 
> void loop(){
>   webSocket.loop();
> }
> 
> no need to use webSocket.beginSSL, no need to use webSocket.begin(https://demo.herokuapp.com) with http or https and no need to provide port number, the above code is working.

Very good to know, so it is posible to use this lib with external server then. Thank You  @Sakthivelgovindan

fakhrilak, 2020-11-26:
> Hi @OmarGC, @elastra21, @Chevindu , @raulest50, @Muzzamil1 , @rishabhrpg and @timum-viw ,
> I tried to connect using the below code, It's working fine
>
> Example :
>
> #include <SocketIoClient.h> [...]

thanks....its worked

closed 2021-01-13 by timum-viw