timum-viw/socket.io-client #53: Doesn't reconnect once disconnected

vicky-gonsalves, 2019-05-16:
Socket Does not reconnect to server once its disconnected. Is there any way we can forcefully reconnect it to server?

alexinasra, 2019-06-23:
Hi, I have the same problem! the Device does not reconnect after disconnecting and the disconnect event does nothing.

```
  socket.on("disconnect", [](const char* payload, size_t len) {
    Serial.println("[MyApp] Disconnected!");
    socket.begin("10.0.0.14", 8181);
  });
  
```

igurug, 2020-05-07:
Have the same issue. Will try to solve it modifying the SocketIoClient.cpp. I don't know the results yet, but I added "ESP.restart();" into this part. 
"switch(type) {
		case WStype_DISCONNECTED:
			SOCKETIOCLIENT_DEBUG("[SIoC] Disconnected!\n");
                        ESP.restart();
			break;"

Now when it loses connection, the device is restarted and tries to connect from zero.
Nevertheless, I suppose the issue is rather in the WebSocketClient

igurug, 2020-05-08:
> Have the same issue. With a twist. I host my own socket.io server. Whem the connection is lost due to WiFi becomes unreachable, it easilly reconnects. However, when I reboot the modem and its external IP is changed, the ESP never reconnects.
> 
> Will try to solve it modifying the SocketIoClient.cpp. I don't know the results yet, but I added "ESP.restart();" into this part.
> "switch(type) {
> case WStype_DISCONNECTED:
> SOCKETIOCLIENT_DEBUG("[SIoC] Disconnected!\n");
> ESP.restart();
> break;"
> 
> Now when it loses connection, the device is restarted and tries to connect from zero.
> Nevertheless, I suppose the issue is rather in the WebSocketClient

Didn't work with ESP.restart(). Worked with ESP.reset()

closed 2021-01-13 by timum-viw