timum-viw/socket.io-client #56: SocketIOClient is eating heap like hungry monster

RobertHolscher, 2019-09-07:
Hi I have succesfully implemented the Socket.io library into my ESP8266. I have noticed that when I am using and initializing the SocketIoClient, the heap is reduced from 20000 to 4500 
When I am then receiving some calls and do some other computation, the heap is dangerously low and causing resets.

Why does it eat away so much heap by default and can this be reduced somewhere in the library ?
Your help is appreciated

lulersoft, 2020-01-14:
when emit message ,a memory leak occurs, it is normal to receive the message，For unknown reasons, please give me some advice。It seems that the message package sent has not been destroyed。

lulersoft, 2020-01-15:
in void SocketIoClient::loop() ,add this code: std::vector<String>(_packets).swap(_packets):

	if(millis() - _lastPing > PING_INTERVAL) {
		_webSocket.sendTXT("2");
		_lastPing = millis();
		std::vector<String>(_packets).swap(_packets);//DC
	}

GowthamGottimukkala, 2020-06-13:
> Hi I have succesfully implemented the Socket.io library into my ESP8266. I have noticed that when I am using and initializing the SocketIoClient, the heap is reduced from 20000 to 4500
> When I am [...]

Hey! I am new to this. How can I check my heap?
Any help is appreciated

closed 2021-01-13 by timum-viw