timum-viw/socket.io-client #60: Cannot emit events to server

JoeToeniskoetter, 2020-01-02:
I cannot seem to catch any events emitted to my nodejs server from the esp. 

The esp will connect to the socket.io server fine but when I call:
    webSocket.emit("temp_update", "{\"temp\":\"35\"}");
and then add an event listener on my server of :
io.on("temp_update", (data) => console.log(data))
my server never receives the event.

coma0815, 2020-01-02:
What is the the output of your serial monitor? this library should print some infos to the serial monitor...

My string to be sent would look like this, so that the sting is not closed until the end:
webSocket.emit("temp_update", "{\"temp\":\"35\"}");

Does this work for you?

JoeToeniskoetter, 2020-01-02:
Hi,
Thank you for your reply. 
I tried replacing my emit function with the one you provided but I am having issues compiling with the following error: 

error: expected ')' before 'temp'
   webSocket.emit("temp_update", "{"temp":"35"}");
                                                                ^

When using the example provided for emitting an event:
webSocket.emit("jsonObject", "{\"foo\":\"bar\"}");

I am getting the following output in the console. 
[SIoC] add packet 42["jsonObject",{"foo":"bar"}
[SIoC] packet "42["jsonObject",{"foo":"bar"}" emitted

But still am not receiving on the server.

coma0815, 2020-01-02:
Oh just realized that my comment was not printed correctly. In front of every “ that needs to be in the json string there is a backslash \ so that the string is not closed before. Anyways - with the foo bar example it should work.. 

Is the connection to the socket server working fine? What is the output of the server console?
 I think my socket server prints a message to the servers console if there is a event, that is not listened to. 

Have you tried sending messages from the server to the esp?

JoeToeniskoetter, 2020-01-02:
Yes the esp does connect to the socketio server successfully, and the esp receives events sent from the server fine as well.

`  io.on('connection', function (socket) {
    console.log('socket id', socket.id);
    socket.emit('event', {msg: 'event'});
  })
}`
in the node server console: 
socket id FmVg9oFI_QUiOUREAAAA

The event sent from the socketio server is also received by the esp
showing the following output in the serial montior:

17:17:44.611 -> [SIoC] trigger event event
17:17:44.611 -> got message: {"msg":"event"}

Here is my entire sketch for the esp. I am using WifiManager to connect to the network.
I hope I'm not missing something obvious. Thanks again for your help!

`#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <WiFiManager.h> 

#include <SocketIoClient.h>

#define USE_SERIAL Serial

WiFiManager wifiManager;
SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
    USE_SERIAL.begin(115200);
    USE_SERIAL.setDebugOutput(true);
    wifiManager.autoConnect("AutoConnectAP");

    webSocket.on("event", event);
    webSocket.begin("192.168.1.13", 8081);
    webSocket.on("connect", connectHandler);
    webSocket.on("disconnect", disconnectHandler);
    webSocket.emit("jsonObject", "{\"foo\":\"bar\"}");
}

void loop() {
    webSocket.loop();
}

void connectHandler(const char * payload, size_t length){
  webSocket.emit("jsonObject", "{\"foo\":\"bar\"}");
}

void disconnectHandler(const char * payload, size_t length){
  Serial.println("Disconnected event worked");
}`

coma0815, 2020-01-02:
My code basically looks the same and I'm also using wifimanager to connect.. 
The only thing I see that looks different from mine is that you "begin" the socket.io connection before you add the event listeners for connect & disconnect.. I think they should come first, but this shouldn´t cause the problem. 

another thing: maybe try to not emit right after you begin the connection - the server might not be ready to receive the message. and the emit in the connectHandler function might not be triggered because afair it should come before the webSocket.begin-call..

so maybe try like this:

```
webSocket.on("event", event);
webSocket.on("connect", connectHandler);
webSocket.on("disconnect", disconnectHandler);
webSocket.begin("192.168.1.13", 8081);

delay(2000);      // not sure if blocking code is the best way to go.. ;)
webSocket.emit("jsonObject", "{\"foo\":\"bar\"}");
webSocket.emit("plainString", "\"this is a plain string\"");      // just to see if it is not something with the json format..
```

JoeToeniskoetter, 2020-01-03:
Your example is still not being received by the server. 
I have no idea what could be going wrong here. 

I removed all listeners except the basic event handler:
```webSocket.on("event", eventHandler);```

and the emit from the esp is still not received on the server.

I suppose I should delete the library and download again from this repo maybe?
I am using socket.io 2.3.0 on the server side, could different versions of the library have an impact?

JoeToeniskoetter, 2020-01-03:
@coma0815 reinstalling the library did not give me anything. Have you had success with any other esp SocketIO libraries? 

Just trying to isolate the issue to the library or my server code.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

coma0815, 2020-01-03:
I just looked it up and I'm using version 2.1 on server side. you could try using this version (https://cdnjs.com/libraries/socket.io/2.1.0) because I think the syntax on server side looks slightly different as there are no "=>" in my code.
Did you try the basic examples like the chat or whiteboard coming with the library just to make sure that this is working correctly?

I never tried an other library on esp because this worked just fine on esp8266 and esp32 and other libraries didn't looks so convenient to me.

JoeToeniskoetter, 2020-01-03:
I tried to remove my arrow functions as well but it didn’t seem to work. 

Fortunately I did get my server to receive events using the arduinoWebSockets library with SocketIO version 2.3.0.

Thanks for your help I will close this issue.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-01-03 by JoeToeniskoetter

Kperroch, 2020-01-26:
@JoeToeniskoetter How did you solve your issue ?

JoeToeniskoetter, 2020-01-26:
> @JoeToeniskoetter How did you solve your issue ?

I switched to a different library for the esp. I am now using the arduinoWebSockets library.

oninsomnus, 2021-04-02:
I had the same issue and I had to switch from socket.io version 4 to 2.3.0 in server side, after that it worked fine.

Xarbenence, 2024-01-20:
> @Kperroch
> 
> I switched to a different library for the esp. I am now using the arduinoWebSockets library.

I know it's been 4 years... but I am currently struggling to catch any events with my ESP32 using this library or the ArduinoWebSockets library. I can successfully ping the server and receive a pong, but when I press a button the server side which should send an event/message out to all clients, I get radio silence. I tried connecting to the server with websocat, and I get the same behavior (not seeing any events when I should). Is this likely an indication of the server not transmitting events/messages correctly? Or did you need to implement something special to be able to receive Socket.io events/messages ? ChatGPT keeps telling me that Socket.io might not be compatible with the AruinoWebSockets library, but I'm not sure that's true.