timum-viw/socket.io-client #61: JSON Emit - How To Include Variables In JSON Portion -> CONCATINATION?

BPLD, 2020-01-02:
I am having troubles including variables to the following code.
My sensor extracts CAN information temperature and level (floats or doubles).

These two values are converted:
dtostrf(tempResult, 5, 1, tempResultCon);
dtostrf(levelResult, 5, 1, levelResultCon);

This "EMIT" sends fine and is well received:
webSocket.emit("incoming data", tempResultCon);

This "EMIT" sends fine and is well received:
webSocket.emit("incoming data", "{\"temp\":\"56\",\"level\":\"85\"}");  

This "EMIT" is giving me the finger no matter what I try :(!
webSocket.emit("incoming data", "{\"temp\":\"" +tempResultCon+ "\",\"level\":\"" +levelResultCon+ "\"}");  

ERROR:
error: invalid operands of types 'const char [10]' and 'char [8]' to binary 'operator+'
webSocket.emit("incoming data", "{\"temp\":\"" +tempResultCon+ "\",\"level\":\"" +levelResultCon+ "\"}");

Any support/help will be highly appreciated :)!
Thank you in advance.

coma0815, 2020-01-02:
I think you cannot combine two char arrays with the operator "+". 
This is at least what the error message says.
There are a lot of solutions to this online - I myself make a std::string where I can combine values and text to json and convert it to a char array (this is was the emitting-function is waiting for) just before emitting. I use something like this: 

```
std::string x = "{\"msgType\":\"2\",\"lmpStat\":\"";
    x.append(itoa(stat, numstr, 10));
    x.append("\"}");

char message[x.length() + 1];
  message[0] = '\0';
  strcpy(message, x.c_str());

webSocket.emit("messagetype", message);
```

This might not be a clean / resource-friendly way to program but it works for me...

BPLD, 2020-01-02:
AWESOME, thanks for the quick reply and quick resolution to my problem!
I modified the code with my data and it worked!
Many thanks!

OUTPUT: [SIoC] add packet 42["incoming data",{"temp":"-40","level":"0"}]
4 thumbs up!

closed 2020-01-02 by BPLD