timum-viw/socket.io-client #64: Can't connect my server

0x587, 2020-01-16:
my mcu code:
[image]

> webSocket.begin("192.168.0.203", 5000);

I use flask server.
[image]

And I use js connect my server success,so ithink my server is working.
[image]

0x587, 2020-02-04:
## HI,anyone there??

3omer, 2020-08-25:
this might help [559](https://github.com/Links2004/arduinoWebSockets/issues/559#issuecomment-680320792)

closed 2021-01-13 by timum-viw