timum-viw/socket.io-client #65: Submit handshake data on connect?

rx2347, 2020-01-30:
First of all: Very useful library, thanks for that.

I was wondering: Is there a way to submit additional handshake data on connect?
Looking for the equivalent to io.connect('localhost', { query: "foo=bar" });

That'd be a nice feature to add in the future, if its not already possible somehow?

sergiodelossantosjr, 2020-04-07:
Hi I already done how to submit handshake data on connect from arduino to nodejs, I hope it helps.

In arduino I use this statement: 

char path[] = "/socket.io/?transport=websocket**&myParameter=MY PARAMETER VALUE HERE**"; // Socket.IO Base Path
webSocket.begin("127.0.0.1", 3000, path);

In nodejs I use code below:

const server = require('http').createServer();
const io = require('socket.io')(server);

io.on('connection', client => {

   
    if(typeof **client.handshake.query.myParameter** != undefined){ //evaluate the query parameter here
	    
                 var _myParameter = **client.handshake.query.myParameter**;  //get the query parameter here
		 
                 console.log("connected");
	         console.log("MY PARAMETER FROM QUERY: "+ _myParameter);
		 

    }
   
});
server.listen(3000);

I think this feature is already here for a long time.

gilphilbert, 2020-04-08:
It depends on how you want to send the traffic. If you want to send it via GET, then you can append the query to include the data (check out the answer from @sergiodelossantosjr above).

If you want to send it as HTTP headers, then you need a different way. I've just opened a PR to do this, but it looks like this project has largely been abandoned. Check out the code in #71 for a way to send additional HTTP headers.

closed 2021-01-13 by timum-viw