timum-viw/socket.io-client #69: Is it support for esp32?

ducthang123, 2020-03-27:
I used to esp32 and I want to esp32 connect realtime with socket io server.  Have this library support for esp32?

coma0815, 2020-03-27:
Yes - should work fine. I use it with esp32 for more than a year now

ducthang123, 2020-04-03:
Thank you so much coma0815

gilphilbert, 2020-04-08:
It works fine - but make sure your use the latest version on GitHub, not the version in the Arduino IDE library manager.

bartbierling, 2020-04-22:
Hi Guys, I am using an ESP32 to connect to a Socket IO server, which works and I am able to connect to it using an Android app. However, when I try to connect to it with the ESP32 I am unable to connect.  When I try to connect I get to the server I get the following error: 
`1|socket-server  | 2020-04-22T11:35:17.828Z worker:server 🚀 Server listening for WebSockets on port 3001
1|socket-server  | Error: read ECONNRESET
1|socket-server  |     at TCP.onStreamRead (internal/stream_base_commons.js:205:27) {
1|socket-server  |   errno: 'ECONNRESET',
1|socket-server  |   code: 'ECONNRESET',
1|socket-server  |   syscall: 'read'
1|socket-server  | }
`

and I get the following error from from the ESP:
`[E][WiFiClient.cpp:260] connect(): socket error on fd 55, errno: 104, "Connection reset by peer"`
It is caused by something in the socket.loop(). Has someone encountered something similar or a suggestion what the cause could be? Thank you in advance!

my code: 

`
char host[] = "myserver";
int port = 3001;
char path[] = "/facility";

SocketIoClient socket;
WiFiClient client;

void event(const char * payload, size_t length) {
  Serial.print("got message! ");
  Serial.println(payload);
}

void setup() {
  Serial.begin(115200);
  delay(500); 
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) { //Check for the connection
    delay(500);
    Serial.println("Conneting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");

  socket.on("serverData", event);
  socket.begin(host, port, path);  
  socket.emit("microcontrollerData", "ESP message!! ");
}

void loop() {
  socket.loop();
  socket.emit("microcontrollerData", "hello from ESP32.1");
  if(WiFi.status() == WL_DISCONNECTED) {
    Serial.println("wifi disconnected");
  }
  delay(1000);  
}`

coma0815, 2020-04-22:
Are you receiving the  
`socket.emit("microcontrollerData", "ESP message!! ");`
message on the server side from your ESPs setup-section?
If no, the error might be occurring already earlier than socket.loop();

Are you sure that your server-side is socket.io and not a normal websocket-server? The can not work together afair...

bartbierling, 2020-04-22:
Thank you for your reaction! 
No, the message is never emitted, actually, the ESP is never actually connected to the server (the server doesn't log the connection of the ESP). Yes, I am sure it is socketIO!

bartbierling, 2020-04-23:
So I've identified the problem; I was trying to connect to the namespace, which creates these errors. When I connect to the default path (so not adding a path only the port and serveraddress it works fine!

closed 2021-01-13 by timum-viw

WellingtonSouzaAbreu, 2021-02-04:
> Sim, deve funcionar bem. Eu uso com esp32 há mais de um ano

Pode me ajudar?

Sempre tenho esse erro: 

C:\Users\welli\Documents\Arduino\libraries\SocketIoClient\SocketIoClient.cpp: In member function 'void SocketIoClient::webSocketEvent(WStype_t, uint8_t*, size_t)':
C:\Users\welli\Documents\Arduino\libraries\SocketIoClient\SocketIoClient.cpp:40:27: error: 'hexdump' was not declared in this scope
    hexdump(payload, length);
                           ^
Foram encontradas múltiplas bibliotecas para "WiFi.h"
Usado: C:\Users\welli\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi
Não usado: C:\Users\welli\Documents\Arduino\libraries\WiFiEspAT
Não usado: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1