timum-viw/socket.io-client #73: Can't connect to server

nvomedia, 2020-04-12:
Hi,

I'm trying to connect the client to a server at my localhost (and if that works to a live server).
I have used Node.JS and Socket.IO for that server and it's working in my browser, now I try to connect to the same server using my Wemos D1 Mini.

I have installed the library, and my code compiles. But I can't connect to my server. I have tried almost every possibility but the only message I get is ``[WSc] Disconnected!``

Based on the code from the example I have now this:

```
webSocket.begin("127.0.0.1", 5000, "/socket.io/?EIO=3&transport=websocket");
webSocket.onEvent(webSocketEvent);
webSocket.setReconnectInterval(5000);
webSocket.enableHeartbeat(15000, 3000, 2);
```
I have also tried ``localhost`` and no path and the default path. My local server is running port 5000 and as I said, that is working in my browser. In my network console from my browser I see that the webservice requests also are sended to that address.

How can I get this to work?

nvomedia, 2020-04-13:
I made a fault in here. From the side of the ESP8266 the host is of course not 127.0.0.1 but the internal IP of the router. 
So that is fixed.

But now I get an `Error: read ECONNRESET` in Node.JS after the ESP is connected. I did a few searches about that, and it looks that that is because an incorrect close of the connection. 
Did someone else had this problem in combination with a Node.JS server?

nvomedia, 2020-04-13:
I used the wrong example... my bad.. I had to use the SocketIO and now I can connect to the server!

closed 2020-04-13 by nvomedia

gonzaloeiza, 2021-05-20:
Hey, how did you managed to solve this problem? I have a similar problem with my node js server. When de ESP is connected an ECONNRESET error appears and I can´t solve it.
Here is my code:
`/*
 * WebSocketClientSocketIO.ino
 *
 *  Created on: 06.06.2016
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <WebSocketsClient.h>

#include <Hash.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;

#define USE_SERIAL Serial

#define MESSAGE_INTERVAL 30000
#define HEARTBEAT_INTERVAL 25000

uint64_t messageTimestamp = 0;
uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void webSocketEvent(WStype_t type, uint8_t * payload, size_t lenght) {

    switch(type) {
        case WStype_DISCONNECTED:
            USE_SERIAL.printf("[WSc] Disconnected!\n");
            isConnected = false;
            break;
        case WStype_CONNECTED:
            {
                USE_SERIAL.printf("[WSc] Connected to url: %s\n",  payload);
                isConnected = true;

			    // send message to server when Connected
                // socket.io upgrade confirmation message (required)
				webSocket.sendTXT("5");
            }
            break;
        case WStype_TEXT:
            USE_SERIAL.printf("[WSc] get text: %s\n", payload);

			// send message to server
			// webSocket.sendTXT("message here");
            break;
        case WStype_BIN:
            USE_SERIAL.printf("[WSc] get binary lenght: %u\n", lenght);
            hexdump(payload, lenght);

            // send data to server
            // webSocket.sendBIN(payload, lenght);
            break;
    }

}

void setup() {
    // USE_SERIAL.begin(921600);
    USE_SERIAL.begin(115200);

    //Serial.setDebugOutput(true);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("XXXXXXXX", "XXXXXXXXX");

    //WiFi.disconnect();
    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    webSocket.begin("192.168.1.76", 8080);
    //webSocket.setAuthorization("user", "Password"); // HTTP Basic Authorization
    webSocket.onEvent(webSocketEvent);

}

void loop() {
    webSocket.loop();

    if(isConnected) {

        uint64_t now = millis();

        if(now - messageTimestamp > MESSAGE_INTERVAL) {
            messageTimestamp = now;
            // example socket.io message with type "messageType" and JSON payload
            webSocket.sendTXT("42[\"messageType\",{\"greeting\":\"hello\"}]");
        }
        if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
            heartbeatTimestamp = now;
            // socket.io heartbeat message
            webSocket.sendTXT("2");
        }
    }
}`