timum-viw/socket.io-client #74: webSocket.on("disconnect", handler) doesn't work

Herant, 2020-05-24:
Hey,

Not sure if i'm doing this wrong but, i'm insterested in knowing when socket.io server is connected to esp8266. So in void setup() i declared:
webSocket.on("connect", ioConnected);
which sets my connection status variable to true, i would also like to have my variable set to false in case i'll lose connection so i thought declaring:
webSocket.on("disconnect", ioDisconnected);
would do the trick, but it literally do nothing.
Any idea why?

coma0815, 2020-05-25:
I think this is part of an open pull request, that is not merged yet.. 

https://github.com/timum-viw/socket.io-client/pull/49 
> When server closes connection the "disconnect" event never is called. Adding that line (33) the event "disconnected" is triggered when server closes the connection, calling the function in the line 125.

You can add this one line of code yourself. I guess then it should work..

@timum-viw Any chance you could merge that PR? It is open for quite some time now.. 🙏

Herant, 2020-05-25:
Ah cool!

Although i managed to add my own feature to watch if device is connected to socket server or not.

Added in SocketIoClient.cpp:

bool io_Status = false;

bool SocketIoClient::status()
{
	return io_Status;
}

Inside case WStype_DISCONNECTED is set io_Status = false, and inside case WStype_CONNECTED is set set io_Status = true.

Also added one line of code inside SocketIoClient.h:

Inside class SocketIOClient, just under "void loop();" i added "bool status();"

So now whenever i need to know if device is connected to socket server, i just check it by running status command inside the loop :

if(socket.status()){
    Serial.println("connected");
  } else {
    Serial.println("disconnected");
  }

Don't know if it's a good or bad solution since i'm new to the whole arduino/cpp thingy, but it works well for my purposes.

nathapob, 2020-06-15:
This way work for me

Step 1 : Go to library Check
you can find destination library folder Go to File > Preference
Step 2 : Go to WebSockets\src > Delete or change name files 
"SocketIoClient.cpp" & "SocketIoClient.h" My way deleted

Step 1 & Step 2  > confirm the correct file #include 
Step 3 : Edit Code Library in folder > SocketIoClient
Step 4 : Open Code  Add Code Uder Line 22 This Code "trigger("disconnect", NULL, 0);" 

```
void SocketIoClient::webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
	String msg;
	switch(type) {
		case WStype_DISCONNECTED:
			SOCKETIOCLIENT_DEBUG("[SIoC] Disconnected!\n");
			trigger("disconnect", NULL, 0);
			break;
		case WStype_CONNECTED:
			SOCKETIOCLIENT_DEBUG("[SIoC] Connected to url: %s\n",  payload);
			break;
		case WStype_TEXT:
			msg = String((char*)payload);
			if(msg.startsWith("42")) {
				trigger(getEventName(msg).c_str(), getEventPayload(msg).c_str(), length);
			} else if(msg.startsWith("2")) {
				_webSocket.sendTXT("3");
			} else if(msg.startsWith("40")) {
				trigger("connect", NULL, 0);
			} else if(msg.startsWith("41")) {
				trigger("disconnect", NULL, 0);
			}
			break;
		case WStype_BIN:
			SOCKETIOCLIENT_DEBUG("[SIoC] get binary length: %u\n", length);
			hexdump(payload, length);
		break;
	}
}
```

Saved File Upload Code to Arduino

This my Arduino Code Example 

```
void on_disconnect(const char * payload, size_t length) {
  // calls on_disonnect of web socket
  Serial.println("Disconnected to Remote Server.");
  isConnected = false;
}

void setup() {

webSocket.on("disconnect", on_disconnect);

}
```

I use in the case **NodeJs.Socket IO Server Down** stop Arduino Socket.IO Client Add Packet Data

Rafdal, 2020-07-28:
That `trigger("disconnect", NULL, 0);` in the `case WStype_DISCONNECTED:` perfectly solved the problem, thanks :)

closed 2021-01-13 by timum-viw