timum-viw/socket.io-client #77: Complie Error

xdubx, 2020-06-11:
home/hp/Arduino/libraries/SocketIoClient/SocketIoClient.cpp:41:27: error: 'hexdump' was not declared in this scope
    hexdump(payload, length);
                           ^
I complied it for esp32. Uncomment it and it work again.

coma0815, 2020-06-12:
Take a look at this issue: https://github.com/timum-viw/socket.io-client/issues/15

xdubx, 2020-06-18:
Okey. Seems to be solved.

closed 2020-06-18 by xdubx

aranchuzz, 2022-09-30:
How did you solve it?

dlowrance, 2023-03-11:
I was able to get it to compile by commenting out this line:
//hexdump(payload, length);

In this file:
C:\Users\YourUsername\Documents\Arduino\libraries\SocketIoClient\SocketIoClient.h

I'm new to C++ but I assume this isn't critical to it's functionality as it looks like a logging utility