timum-viw/socket.io-client #78: Support for rooms and namespaces

GowthamGottimukkala, 2020-06-13:
I am using this library for 1000 nodemcus (each nodemcu acts as a client)
How to add support for **rooms** and namespaces so that, from the server I can emit messages to only a set of nodemcus(say 200 of 1000)

Any help appreciated. I'm new to this

coma0815, 2020-06-15:
There has been an issue about this just 2 weeks ago. 
While namespaces are not supported by this library, there seems to be a workaround discussed in the link that I posted in the other issue: https://github.com/timum-viw/socket.io-client/issues/76

Namespaces are also not really supported by the library that this one here is based on but there was an attempt to implement it. 
I guess the link above and the link posted there will help.

If you accomplish to make a good working implementation, it would be great if you could open up a pull request or at post your working code here!
Thank you!

GowthamGottimukkala, 2020-06-16:
> There has been an issue about this just 2 weeks ago.
> While namespaces are not supported by this library, there seems to be a workaround discussed in the link that I posted in the other issue: #76 [...]

Thank you

closed 2020-06-16 by GowthamGottimukkala