timum-viw/socket.io-client #82: [SIoC] Disconnected! on Connection

Stephan-MDD, 2020-07-23:
Hi, Im getting a `[SIoC] Disconnected!` message when I run my program.  
My server socket is connected to a react application so I know the server works.  
But when connecting with my ESP8266 nothing appears in my node logs.

Hope some of you can help me :)

**CPP ESP8266 Controller**
```cpp
// libraries
#include <ESP8266WiFi.h>
#include <SocketIoClient.h>

// definitions
#define print(value) Serial.println(value)

SocketIoClient socket;

void setup() {
    WiFi.begin("my ssid", "my password");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        print("*");
    }

    print("WiFi connection Successful");
    print(WiFi.localIP());

    socket.begin("http://localhost:4000/", 4000);
}

void loop() {
    socket.loop();
}
```

**Node Server**
```javascript
const socket = require('socket.io');

const connect = (server) => {
	const io = socket(server);
	io.on('connection', (socket) => console.log('made socket connection', socket.id));
};
```

CPP Serial Monitor Output
```bash
******
WiFi connection Successful
My IP Address
[SIoC] Disconnected!
[SIoC] Disconnected!
...
```

GowthamGottimukkala, 2020-07-25:
Replace the localhost with you server's ip and it should work.

Stephan-MDD, 2020-07-25:
I tried using NGROK to tunnel my localhost but got the same error :(

LittleB0b, 2020-07-31:
Remove the "http://" and it should work

Stephan-MDD, 2020-07-31:
Thanks, I used MQTT over Websockets instead :)

closed 2021-01-13 by timum-viw

motionharvest, 2021-12-07:
I know this is late, but your appending the port to the url and then passing it as the second argument to begin()...