timum-viw/socket.io-client #86: Error compiling for board

Cage-Y, 2020-10-05:
Hi, I'm new with the ESP-board and until now all of the example sketches for ESP8266 have worked. I really would like to work with socket.io and I'm really looking forward coding with this little piece of technology :D. I'm using an ESP-12E. 

I've been working on this library for hours now and I tried a lot of different things. But I can't get around the error "compiling for board". I added the "-lstdc++" in the platform.txt file, and I thought that would fix it... But it didn't. I've read the other issues, but I couldn't find one helping me. I'm using the example provided (I changed SSID & Password to what it was):

```#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <SocketIoClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;
SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
    USE_SERIAL.begin(115200);

    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("SSID", "passpasspass");

    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    webSocket.on("event", event);
    webSocket.begin("my.socket-io.server");
    // use HTTP Basic Authorization this is optional remove if not needed
    webSocket.setAuthorization("username", "password");
}

void loop() {
    webSocket.loop();
```
And I got this as an error:

```
Arduino: 1.8.13 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\Gebruiker\Documents\Arduino\libraries\SocketIoClient\SocketIoClient.cpp: In member function 'void SocketIoClient::beginSSL(const char*, int, const char*, const char*)':

C:\Users\Gebruiker\Documents\Arduino\libraries\SocketIoClient\SocketIoClient.cpp:47:50: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

  _webSocket.beginSSL(host, port, url, fingerprint);

                                                  ^

In file included from C:\Users\Gebruiker\Documents\Arduino\libraries\SocketIoClient/SocketIoClient.h:7:0,

                 from C:\Users\Gebruiker\Documents\Arduino\libraries\SocketIoClient\SocketIoClient.cpp:1:

C:\Users\Gebruiker\Documents\Arduino\libraries\WebSockets\src/WebSocketsClient.h:50:10: error:   initializing argument 4 of 'void WebSocketsClient::beginSSL(const char*, uint16_t, const char*, const uint8_t*, const char*)' [-fpermissive]

     void beginSSL(const char * host, uint16_t port, const char * url = "/", const uint8_t * fingerprint = NULL, const char * protocol = "arduino");

          ^

exit status 1

Error compiling for board Generic ESP8266 Module.
```

I would really appreciate someones help!

kurtbeheydt, 2020-10-13:
I have the same error. Created a fix and pull request: https://github.com/timum-viw/socket.io-client/pull/88

tuananhzippy, 2020-10-14:
I have the same error. Please fix this soon.

coma0815, 2020-10-14:
Not sure if I can help - but does this error just pop up for a specific board or would you have the same error if you choose another board? 

this library is built on another socket library, that is constantly updated, while this library is not being updated for years. So if the error appears for all boards I would say that it is not compatible anymore with the base-library...
For a quick fix try to install an older version from the socket library to match it again with this socket-io library.. but this is just a guess..

ankugit, 2020-10-14:
@kurtbeheydt The fix worked for me

tuananhzippy, 2020-10-14:
> @kurtbeheydt The fix worked for me

Awesome, this library works. Thank for sharing.

Cage-Y, 2020-10-28:
@kurtbeheydt Took me a long time :P But it finally worked! Thanks a lot!

lucalas, 2020-11-08:
News about it?

closed 2021-01-13 by timum-viw