timum-viw/socket.io-client #91: Socket.io update V3 does not work with the library. protocol change

fer3412, 2020-11-25:
Socket.io V2.x work fine
```
`{
    "name": "esp32_socketio",
    "version": "1.0.0",
    "description": "socket.io app",
    "dependencies": {
        "express": "^4.17.1",
        "socket.io": "^2.2.0"
    },
    "scripts": {
        "start": "node index.js"
    }
}`
```

Socket.io V3 does not work

```
`{
    "name": "esp32_socketio",
    "version": "1.0.0",
    "description": "socket.io app",
    "dependencies": {
        "express": "^4.17.1",
        "socket.io": "^3.0.1"
    },
    "scripts": {
        "start": "node index.js"
    }
}`
```

**any idea to try to update the library**
https://socket.io/docs/v3/migrating-from-2-x-to-3-0/index.html

faizanjaffer, 2020-12-01:
Same issue faced by me...

danielmweb, 2021-01-03:
Thanks for this insight, i was trying to get this to work with V3 but no success. Changed to v2 and now it works fine.

closed 2021-01-13 by timum-viw