timum-viw/socket.io-client #96: Compilation Error.

rodentskie, 2021-01-09:
Good day,

I used the example of WebSockets by Markus Sattler the WebSocketClientSocketIO and yes it can connect to my server.

[image]

But when I'm using the example from this library, I can't even compile my code. Here is the error I get.

[image]

I comment and uncomment some stuff but all of this is from the example.
Please help Sir / Ma'am.

`
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <SocketIoClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;
SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
  USE_SERIAL.begin(115200);

  USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFiMulti.addAP("OnePiece", "dFGSJZ8476");

  while (WiFiMulti.run() != WL_CONNECTED) {
    delay(100);
  }
//
//
//  webSocket.begin("192.168.1.34", 5000);
//  webSocket.on("event", event);
  // use HTTP Basic Authorization this is optional remove if not needed
  //  webSocket.setAuthorization("username", "password");
}

void loop() {
  webSocket.loop();
}
`

Here is the code.

closed 2021-01-13 by timum-viw

kevinlhoste, 2021-04-02:
hi , 
I got exctly the same error how did you solve it ?
Thanks

rodentskie, 2021-04-02:
> hi ,
> I got exctly the same error how did you solve it ?
> Thanks

Go to the library's .cpp file and remove the beginSSL block of code. That's how I fixed the issue.