timum-viw/socket.io-client #101: argument error on WebSocketsClient::beginSSL. with the default example.

z9fr, 2022-05-01:
Hi, i been trying to connect to a socket.io server using this library. and im facing a issue. below is the complete stack trace

```bash
Arduino: 1.8.19 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/home/dasith/Arduino/libraries/SocketIoClient/SocketIoClient.cpp: In member function 'void SocketIoClient::beginSSL(const char*, int, const char*, const char*)':
/home/dasith/Arduino/libraries/SocketIoClient/SocketIoClient.cpp:46:39: error: invalid conversion from 'const char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
   46 |  _webSocket.beginSSL(host, port, url, fingerprint);
      |                                       ^~~~~~~~~~~
      |                                       |
      |                                       const char*
In file included from /home/dasith/Arduino/libraries/SocketIoClient/SocketIoClient.h:7,
                 from /home/dasith/Arduino/libraries/SocketIoClient/SocketIoClient.cpp:1:
/home/dasith/Arduino/libraries/WebSockets-2.3.6/src/WebSocketsClient.h:50:93: note:   initializing argument 4 of 'void WebSocketsClient::beginSSL(const char*, uint16_t, const char*, const uint8_t*, const char*)'
   50 |     void beginSSL(const char * host, uint16_t port, const char * url = "/", const uint8_t * fingerprint = NULL, const char * protocol = "arduino");
      |                                                                                             ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```
it seems like a issue with  conversion and my code is almost the default example. i have attached it below. 

```cpp
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <SocketIoClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;
SocketIoClient webSocket;

void event(const char * payload, size_t length) {
  USE_SERIAL.printf("got message: %s\n", payload);
}

void setup() {
    USE_SERIAL.begin(115200);

    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("SSID", "passpasspass");

    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    webSocket.on("event", event);
    webSocket.begin("my.socket-io.server");
}

void loop() {
    webSocket.loop();
}
```

thanks in advance.

z9fr, 2022-06-14:
> It happened to me. After some research i realized it has something to do with the board manager
> 
> In the menu bar, open Tools > Board > Boards Manager.
> then type esp8 in the search box
> then from the drop-down menu chose version 2.4.0, then click install (higher board manager versions might work too but i haven't tried it yet)
> 
> then you should be able to compile without errors
> 
> [image]
> 

oh thanks that works !

itaygenkin, 2022-07-08:
I have version 3.0.2, and I had the same issue. So I added casting to fingerprint in the function beginSSL (const char*) --> (uint8_t*), and then everything was working well.

itaygenkin, 2022-07-22:
> > I have version 3.0.2, and I had the same issue. So I added casting to fingerprint in the function beginSSL (const char*) --> (uint8_t*), and then everything was working well.
> 
> Can you give me detail how to do it?

Open  "...Arduino/libraries/SocketIoClient/SocketIoClient.cpp" in your computer, then go to the function "void SocketIoClient::beginSSL(const char* host, const int port, const char* url, const char* fingerprint)" and add the bolding to the following line:
_webSocket.beginSSL(host, port, URL, **(uint8_t*)** fingerprint);

Let me know if you still have a problem.

Jayasubin, 2022-07-26:
I was scratching my head for the past two hours thinking why .beginSSL() was showing up even though I am just using the .begin().

Thanks @itaygenkin for the solution; It worked.