timum-viw/socket.io-client #102: ESP32 is unable to connect to my Node.js socket.io Server

shubhamGhosh-dev, 2023-03-24:
I have uploaded the [example ](https://github.com/timum-viw/socket.io-client/blob/master/examples/BasicExample/ESP32BasicExample.ino)code to my ESP32 board and I found an unusual behavior.

1. When I tried to connect my esp32 to my node.js server where the Socket.Io version is 4.* at that time serial monitor printed `[SIoC] Disconnected!` ( means ESP32 board is unable to connect to my server.).

2. When I tried to connect my esp32 to my node.js server where the Socket.Io version is 2.* or 3.* then also I am getting `[SIoC] Disconnected!`. but at this time my server is also crashing when my ESP32 board is trying to connect ( including when I press the reset button on the ESP32 board). My server is giving this below error.
```
events.js:352
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
Emitted 'error' event on Socket instance at:
    at emitErrorNT (internal/streams/destroy.js:106:8)
    at emitErrorCloseNT (internal/streams/destroy.js:74:3)
    at processTicksAndRejections (internal/process/task_queues.js:82:21) {
  errno: -4077,
  code: 'ECONNRESET',
  syscall: 'read'
}
```
    please let me know what can i do about this or if there is any server specification that I have to take care of.

Lautaro124, 2023-10-25:
Is the position of your node application the same as the one indicated in the example?

MAnonInfinity, 2023-11-06:
@shubhamGhosh-dev The same problem occurs with me too, it keeps on saying disconnected. Were you able to find any other solution?
@Lautaro124 Yes, the node application is the same as in the example, yet I'm facing this issue

Mau-MD, 2024-04-14:
Add `allowEI03` to Socket IO Server configuration
```ts
const io = new Server(server, {
  allowEIO3: true,
  cors: {
    origin: "*",
    methods: ["GET", "POST"],
  },
});
```

edcelmanuel, 2024-10-18:
Try 
```js
socket.begin(server, port, "/socket.io/?EIO=4&transport=websocket");
```