timum-viw/socket.io-client #103: webSocket.emit() triggered in Arduino when I read my RFID but that emit will not received in Node Js

Sakthivelgovindan, 2023-04-20:
### Arduino Code: 

#include <SocketIoClient.h>
#include <ArduinoJson.h>
#include <Arduino.h>
#include <ESP8266WiFi.h>

const char* ssid = "**********";
const char* pass = "*********";

int count = 0;
char card_no[12];

SocketIoClient webSocket;

// From Server Emit Event
void fromNodeServer(const char* message, size_t length){
  Serial.print("Message - ");
  Serial.println(message);

  String cardNo = "5500B465D753";
  webSocket.emit("cardId",("{\"id\":\""+ cardNo +"\"}").c_str());
}

void searchWiFi(){
  int numberOfNetwork = WiFi.scanNetworks();
  Serial.println("----");
  for(int i = 0; i < numberOfNetwork; i++ ){
    Serial.print("Network name: ");
    Serial.println(WiFi.SSID(i));
    Serial.print("Signal strength: ");
    Serial.println(WiFi.RSSI(i));
    Serial.println("--------------");
  }
}

void connectWiFi(){
  WiFi.begin(ssid, pass);
  while(WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    delay(1000);
  }
  Serial.print("");
  Serial.println("WiFi connected");
  Serial.print("IP Address : ");
  Serial.println(WiFi.localIP());
}

void detectRfidTag(){
   if(Serial.available())
   {
      count = 0;
      while(Serial.available() && count < 12)
      {
         card_no[count] = Serial.read();
         count++;
         delay(5);
      }

      String cardNo = String(card_no);
      Serial.print("RFID Card Id : ");
      Serial.println(cardNo);
      webSocket.emit("cardId",("{\"id\":\""+ cardNo +"\"}").c_str()); 
   }
}

void setup()
{
   Serial.begin(9600);
   Serial.setDebugOutput(true);
   searchWiFi();
   connectWiFi();
  
   webSocket.begin("192.168.0.103", 5000);
   webSocket.on("message", fromNodeServer);
}

void loop()
{
 detectRfidTag();
 webSocket.loop();
}

### Node Js Code :

var express = require("express");

const app = express();
const server = require("http").createServer(app);
const io = require("socket.io")(server);

app.get("/", (req, res) => {
  res.send("<h1>Hello world</h1>");
});

io.on("connect", (socket) => {
  console.log(`${socket.id} "connect"`);
  io.emit("message", socket.id);

  socket.on("cardId", function (data) {
    console.log("Card Id", data);
  });

  socket.on("disconnect", () => {
    console.log(`${socket.id} "disconnect"`);
  });
});

server.listen(5000, function () {
  console.log("listening on *:" + 5000);
});

[image]
[image]

valentinsacco, 2023-06-23:
I've the same problem, could you solve it?

timum-viw, 2023-06-25:
hey guys. This library is actually kind of deprecated. Just use https://github.com/Links2004/arduinoWebSockets. It has a SocketIOclient Class.

Sakthivelgovindan, 2024-03-25:
Ok, thank you.

closed 2024-03-25 by Sakthivelgovindan